freifunk-berlin/firmware #814: freifunk-berlin-lib-guard: wrong syntax in makefile?

PolynomialDivision, 2020-06-20:
`/lib/functions/guard.sh` is not available after compiling the package with imagebuilder and installing it.

I think you can not do this:
https://github.com/freifunk-berlin/firmware-packages/blob/master/addons/freifunk-berlin-lib-guard/Makefile#L31

PolynomialDivision, 2020-06-20:
Sry, I think this should work, but for me it is not working. :/

PolynomialDivision, 2020-06-20:
Finally I know why the package has problems:
https://github.com/freifunk-berlin/firmware-packages/blob/master/addons/freifunk-berlin-lib-guard/Makefile#L6

Just replacing this with `include $(INCLUDE_DIR)/package.mk` works! :)

SvenRoederer, 2020-06-20:
Just did a ```curl freifunk-berlin-lib-guard_git-19.359.40465-6e5046b-2_all.ipk |tar -xz ./data.tar.gz --to-stdout | tar -tzv```

for 19.07 (Firmware: git branch "upstream-1907", revision 99d94ca) it resulted in 
```
drwxr-xr-x root/root         0 2019-09-06 10:29 ./
drwxr-xr-x root/root         0 2019-09-06 10:29 ./lib/
drwxr-xr-x root/root         0 2019-09-06 10:29 ./lib/functions/
-rw-r--r-- root/root       656 2019-09-06 10:29 ./lib/functions/guard.sh
```

for openwrt-master (Firmware: git branch "upstream-master", revision b63ae31) result was:
```
drwxr-xr-x root/root         0 2019-09-06 10:29 ./
```

Indeed OpenWrt-master behaves different and produces an empty package.

> Just replacing this with include $(INCLUDE_DIR)/package.mk works! :)

But this skips a lot of shorthand definitions that are done in freifunk-generic.mk and luci.mk

SvenRoederer, 2020-06-20:
probably there is something bad in https://github.com/freifunk-berlin/firmware/blob/daily/upstream-master/patches/packages/luci/0202-split-luci-base_PR2817.patch which causes this

PolynomialDivision, 2020-06-21:
> probably there is something bad in https://github.com/freifunk-berlin/firmware/blob/daily/upstream-master/patches/packages/luci/0202-split-luci-base_PR2817.patch which causes this

I'm currently debugging the Makefile. All includes work. Can we maybe push a temporary workaround? I will stay with your `luci.mk-search` solution, but maybe we switch to official `package.mk` include for all non luci-relevant stuff?

SvenRoederer, 2020-06-21:
I suggest:
* getting https://github.com/openwrt/luci/pull/2637 merged 
  * temporarly use: https://github.com/SvenRoederer/freifunk-berlin-firmware/tree/next/patches/packages/luci/0008* (up to 0011*)
* merging https://github.com/SvenRoederer/freifunk-berlin-firmware/tree/next/patches/packages/packages_berlin/0003* (up to 0008*)

currently this is full functional (http://77.245.36.54:8000/firmwares/unstable/ramips-mt7620/)

PolynomialDivision, 2020-06-21:
> I suggest:
> 
>     * getting [openwrt/luci#2637](https://github.com/openwrt/luci/pull/2637) merged
>       
>       * temporarly use: https://github.com/SvenRoederer/freifunk-berlin-firmware/tree/next/patches/packages/luci/0008* (up to 0011*)
> 
>     * merging https://github.com/SvenRoederer/freifunk-berlin-firmware/tree/next/patches/packages/packages_berlin/0003* (up to 0008*)
> 
> 
> currently this is full functional (http://77.245.36.54:8000/firmwares/unstable/ramips-mt7620/)

I can not access your links?

SvenRoederer, 2020-06-21:
> I suggest:
> 
>     * getting [openwrt/luci#2637](https://github.com/openwrt/luci/pull/2637) merged
>       * temporarly use: https://github.com/SvenRoederer/freifunk-berlin-firmware/tree/next/patches/packages/luci/0008* (up to 0011*)
* https://github.com/SvenRoederer/freifunk-berlin-firmware/blob/next/patches/packages/luci/0008-* luci.mk-reorder-to-have-common-things-in-a-block.patch
* https://github.com/SvenRoederer/freifunk-berlin-firmware/blob/next/patches/packages/luci/0009-luci.mk-reoder-generic-macros-and-add-some-comments.patch
* https://github.com/SvenRoederer/freifunk-berlin-firmware/blob/next/patches/packages/luci/0010-luci.mk-separate-LUCI_BUILD_DEPENDS-from-hardcoded-dependencies.patch
* https://github.com/SvenRoederer/freifunk-berlin-firmware/blob/next/patches/packages/luci/0011-luci.mk-split-into-luci-common.mk-and-luci.mk.patch

>     * merging https://github.com/SvenRoederer/freifunk-berlin-firmware/tree/next/patches/packages/packages_berlin/0003* (up to 0008*)
* https://github.com/SvenRoederer/freifunk-berlin-firmware/blob/next/patches/packages/packages_berlin/0003-freifunk-berlin-generic.mk-use-luci-common.mk.patch
* https://github.com/SvenRoederer/freifunk-berlin-firmware/blob/next/patches/packages/packages_berlin/0004-freifunk-berlin-generic-import-more-required-definitions-after-split-of-luci.mk.patch
* https://github.com/SvenRoederer/freifunk-berlin-firmware/blob/next/patches/packages/packages_berlin/0005-network-defautls-remove-prepare-configure-which-is-defined-via-generic.mk.patch
* https://github.com/SvenRoederer/freifunk-berlin-firmware/blob/next/patches/packages/packages_berlin/0006-freifunk-berlin-generic.mk-remove-SUBMENU-from-package-definiton.patch

being extra verbose ....

PolynomialDivision, 2020-06-21:
Sry, I'm overwhelmed of this number of patches. Why not putting this upstream?
I'm afraid that this makes things complicated.

SvenRoederer, 2020-08-19:
Most of the mentioned patches have been accepted upstream (https://github.com/openwrt/luci/pull/2817#issuecomment-660873641; https://github.com/openwrt/luci/commit/3b2a1e9e1129e77c4ebd5e7ace35653969f2d515#diff-2d8a7a9b9acd99058ff6f4d496ea6ebe) and only 2 patches pending as https://github.com/openwrt/luci/pull/2637.

closed 2020-10-13 by PolynomialDivision