freifunk-berlin/firmware #815: freifunk-berlin-generic: violating PKG_VERSION?

PolynomialDivision, 2020-06-21:
I think setting the `PKG_VERSION` as commit version of the last commit from the feed, is violating the OpenWrt rules that say:
> PKG_VERSION - The upstream version number that we're downloading
> https://openwrt.org/docs/guide-developer/packages

Actually, there is no real upstream that contains the code. It is more like, we should set the `PKG_VERSION:=1` and increase the `PKG_RELEASE`. Seems also weird. :S
https://github.com/freifunk-berlin/firmware-packages/blob/master/freifunk-berlin-generic.mk#L18-L33

SvenRoederer, 2020-06-21:
Our definitions copy a lot from the way done in the LuCI-feed, 
* PKG_SRC_VERSION: https://github.com/openwrt/luci/blob/b39754907ce904747bc03cd5f7b08d682f8fc5e3/luci.mk#L93
* finrev-macro: https://github.com/openwrt/luci/blob/b39754907ce904747bc03cd5f7b08d682f8fc5e3/luci.mk#L68
* package/VERSION: https://github.com/openwrt/luci/blob/b39754907ce904747bc03cd5f7b08d682f8fc5e3/luci.mk#L124

all this up in the same package-definition that we use: https://downloads.openwrt.org/snapshots/packages/x86_64/luci/

So, I'm ok with doing it this way.

SvenRoederer, 2020-06-21:
Also see the 2nd paragraph of the PR description of https://github.com/freifunk-berlin/firmware-packages/pull/188. Merge-conflicts have been most often caused by Version-bumps or version-bumps have not been done at all.

closed 2020-10-13 by PolynomialDivision