freifunk-berlin/firmware #816: luci-lib-freifunk-berlin-wizard: insufficient permissions for uci access

PolynomialDivision, 2020-06-23:
We have to add here a file, otherwise the wizard is not able to access the `/etc/config/profiles_*`.
https://github.com/freifunk-berlin/firmware-packages/tree/master/utils/luci-app-ffwizard-berlin/root/usr/share/rpcd/acl.d

PolynomialDivision, 2020-06-23:
I think this is just wrong:
https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/luci-app-ffwizard-berlin/luasrc/model/cbi/freifunk/assistent/generalInfo.lua#L23

It has to be
```lua
local name = uci:get("profile_"..n, "profile", "name") or "?"
```

PolynomialDivision, 2020-06-23:
Sometimes the ffwizard is accessing config files via the `type` and sometimes via the `name`.
Here is an example
```
config 'community' 'profile'
```
- Type: `community`
- Name: `profile`

Now if we do
```lua
local name = uci:get("profile_"..n, "profile", "name") or "?"
```
we access the name of the community via the `name`.

If we do
```lua
local name = uci:get_first("profile_"..n, "community", "name") or "?"
```
we access the name of the community via the `type`. Since there could be more than one section of the type community (so we get a list back), we need to do `get_first` instead of `get`.

Making things very complicated somtimes the type is called `community` and sometimes the type is used as [name](https://github.com/freifunk/openwrt-packages/blob/master/contrib/package/freifunk-common/files-common/etc/config/freifunk#L11).

PolynomialDivision, 2020-06-23:
Suddenly it works again. I'm not sure why.

PolynomialDivision, 2020-06-23:
We have to switch to
```lua
local name = uci:get("profile_"..n, "profile", "name")
```
and add permissions (this is superuser permission, we have figure out what permissions are actually needed) in `/usr/share/rpcd/acl.d/luci-app-ffwizard-berlin.json` 
```json
{
        "luci-app-ffwizard-berlin": {
                "description": "Super user access role",
                "read": {
                        "ubus": {
                                "*": [ "*" ]
                        },
                        "uci": [ "*" ],
                        "file": {
                                "*": ["*"]
                        }
                },
                "write": {
                        "ubus": {
                                "*": [ "*" ]
                        },
                        "uci": [ "*" ],
                        "file": {
                                "*": ["*"]
                        },
                        "cgi-io": ["*"]
                }
        }
}
```

SvenRoederer, 2020-06-23:
> Sometimes the ffwizard is accessing config files via the `type` and sometimes via the `name`.
> Here is an example

To be honest, all this was confusing me too - always.

SvenRoederer, 2020-06-23:
> We have to add here a file, otherwise the wizard is not able to access the `/etc/config/profiles_*`.
> https://github.com/freifunk-berlin/firmware-packages/tree/master/utils/luci-app-ffwizard-berlin/root/usr/share/rpcd/acl.d

A bit more verbose problem description would be helpful (for others) to reproduce / verify the problem. I personally have not ran the wizard for some long time. I suggest to use the Bug-template.

I assume the problem you have seen is very similar to https://github.com/freifunk/openwrt-packages/issues/23. The solution provided by you seem to relate to the OpenWrt-Forum thread referenced by https://github.com/freifunk/openwrt-packages/issues/23#issuecomment-623022791.

PolynomialDivision, 2020-06-24:
> A bit more verbose problem description would be helpful (for others) to reproduce / verify the problem. I personally have not ran the wizard for some long time. I suggest to use the Bug-template.

I added some useful information and code to debug the luci app.

> I assume the problem you have seen is very similar to freifunk/openwrt-packages#23. The solution provided by you seem to relate to the OpenWrt-Forum thread referenced by freifunk/openwrt-packages#23 (comment).

I think my solution in the PR is a better way? :) For me it seems more secure.

SvenRoederer, 2020-06-24:
> I think my solution in the PR is a better way? :) For me it seems more secure.

Sure, the libuci-lua way might not be able to update uci-settings, as the frontend is not running as root. Also having one lib for reading uci-values and another to write the files via rpc will become really confusing.

closed 2020-10-13 by PolynomialDivision