freifunk-berlin/firmware #817: luci-app

PolynomialDivision, 2020-06-24:
### Bug report

**What is the problem?**

**What is the expected behaviour?**

**Firmware Version:**

**Site Configuration:**

**Custom patches:**

PolynomialDivision, 2020-06-24:
Related:
https://github.com/freifunk-berlin/firmware/issues/814

SvenRoederer, 2020-06-24:
As a sidenote: the issue is not related to the imagebuilder itself, rather than building the packages at all.

The oldest file made by the buildbot (dated 2.June 2020) also shows this problem. So it's hard to tell when it came up. It affects all files based on OpenWrt-master, which use the freifunk-berlin-generic.mk helper. Not sure what changed upstream, but I normally use PR https://github.com/openwrt/luci/pull/2637 for my build, which are not affected by this issue.
So a quick fix is that you use this PR also, as this is the direction we are heading towards.

SvenRoederer, 2020-08-19:
I updated https://github.com/openwrt/luci/pull/2637 an looking forward to get it merged in openwrt-20.x

closed 2020-10-13 by PolynomialDivision