freifunk-berlin/firmware #823: luci-mod-freifunk-ui: footer(?) broken

PolynomialDivision, 2020-06-25:
The footer in the right bottom corner is broken

        Freifunk | Administration

If you click on Freifunk you will receive an error
```
/usr/lib/lua/luci/dispatcher.lua:1189: attempt to index local 'f' (a nil value)
stack traceback:
	/usr/lib/lua/luci/dispatcher.lua:1189: in function 'createtree_json'
	/usr/lib/lua/luci/dispatcher.lua:702: in function 'menu_json'
	/usr/lib/lua/luci/dispatcher.lua:795: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:994: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:478: in function </usr/lib/lua/luci/dispatcher.lua:477>
```
I think here is the code: https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/luci-mod-freifunk-ui/luasrc/controller/freifunk/freifunk.lua
But if I modify it, nothing changes. Any idea which code I have to modify?

The link points to `cgi-bin/luci/freifunk`.

SvenRoederer, 2020-06-25:
This also relates to OpenWrt-master and is caused by some upstream-change. 
We need a LuCI expert here.

Btw. as you might not aware: this package is a 99% copy of  luci-mod-freifunk from  https://github.com/freifunk/openwrt-packages. So it is also affected.

PolynomialDivision, 2020-06-25:
> Btw. as you might not aware: this package is a 99% copy of luci-mod-freifunk from https://github.com/freifunk/openwrt-packages. So it is also affected.

Why?

SvenRoederer, 2020-06-25:
There is a PR / commit-message describing it. If I'm right, the difference is only 3 lines.

PolynomialDivision, 2020-06-25:
> There is a PR / commit-message describing it. If I'm right, the difference is only 3 lines.

I can not find any commit in the history of this package that describes why that is necessary?

SvenRoederer, 2020-06-25:
[image]

PR #80 is the hint  ....
before we applied this patch: https://github.com/freifunk-berlin/firmware/blob/v0.1.2/patches/014-ffwizard-redirect-on-firstboot.patch

SvenRoederer, 2020-06-30:
I opened a corresponding issue "upstream": https://github.com/freifunk/openwrt-packages/issues/33

SvenRoederer, 2020-06-30:
On the Map I found 2 nodes running "Freifunk Berlin Next / 3cf60b8" (sadly does not relate to any known commit) which runs "Powered by LuCI branch (git-20.102.69976-65d673f)" and seem not affected. The relating LuCI-Commit seems https://github.com/openwrt/luci/commit/65d673f from 11. April 2020.

SvenRoederer, 2020-06-30:
> On the Map I found 2 nodes running "Freifunk Berlin Next / 3cf60b8"

@pmelange can you provide more infos?

pmelange, 2020-07-02:
What kind of information do you want? The firmware is based on a forked development branch of openwrt which has since then been merged into openwrt master.

pmelange, 2020-07-02:
```
OPENWRT_REPO=https://github.com/vDorst/openwrt.git
OPENWRT_COMMIT=e4485101ea7ab2c5567ed72fa2e26f15fefa9b82
OPENWRT_BRANCH=ramips-5.4+fixes

PACKAGES_PACKAGES_REPO=https://github.com/openwrt/packages.git
PACKAGES_PACKAGES_COMMIT=1d8d456a55a3135de59731640c58d4c222738036
PACKAGES_PACKAGES_BRANCH=master

PACKAGES_LUCI_REPO=https://github.com/openwrt/luci.git
PACKAGES_LUCI_COMMIT=e523bfd879f34ee92824e97bf57ceead3ac7bc37
PACKAGES_LUCI_BRANCH=master

PACKAGES_FREIFUNK_REPO=https://github.com/freifunk/openwrt-packages.git
PACKAGES_FREIFUNK_COMMIT=f6f61274e0ad2455792858994a79aaa0d3e13b4b
PACKAGES_FREIFUNK_BRANCH=master

PACKAGES_ROUTING_REPO=https://github.com/openwrt-routing/packages.git
PACKAGES_ROUTING_COMMIT=512240f48028bca1832fc79e38f3b40ba9f3b3af
PACKAGES_ROUTING_BRANCH=master

PACKAGES_PACKAGES_BERLIN_REPO=https://github.com/freifunk-berlin/firmware-packages.git
PACKAGES_PACKAGES_BERLIN_COMMIT=a62ea4662475195c0dbe0616103da7be5537a628
PACKAGES_PACKAGES_BERLIN_BRANCH=daily/upstream-master

PACKAGES_PACKAGES_GLUON_REPO=https://github.com/freifunk-gluon/packages.git
PACKAGES_PACKAGES_GLUON_COMMIT=033401c18ecbf5073f5330b4462af6e450e358ff

GLUON_FEEDS='packages luci freifunk routing packages_berlin packages_gluon'
```

SvenRoederer, 2020-07-02:
So we can say, somewhere between https://github.com/openwrt/luci/commit/e523bfd879f34ee92824e97bf57ceead3ac7bc37 and https://github.com/openwrt/luci/commit/09a3b4730a769c43067ff1fb3e4aad339271fbc3 the change was introduced.

closed 2020-10-13 by PolynomialDivision