freifunk-berlin/firmware #826: packages/configtools: "sharenet-switch" should generate a "wan" section

SvenRoederer, 2020-06-28:
The script to swap the assignment (Wan <--> DHCP) of the PoE-port works for 2-port devices. To make it work for 1-Port devices (e.g. NSM Loco, CPE v2+) we need a solid way to generate a "wan" network-section, as this is not done by OpenWrt.

I'm not sure if we should generate a "hardcoded" template wicht we might need to update according to further changes of OpenWrt or if we can "adapt the OpenWrt generate-config" logic.

SvenRoederer, 2020-06-28:
creating the "wan" section manually is a process that can cause some trouble, see: https://github.com/freifunk-berlin/firmware-packages/commit/24bd7d1de2a3852fb7450f2984b3af404f2191e5

foudfou, 2020-07-04:
@SvenRoederer any hints on creating the "section" manually ? Possibly with appropriate firewall zones/rules ?
So i tried to remove `eth0` from the `freifunk` zone, and create a static network called "WAN" on eth0. But wifi clients still can't reach the internet. I'll check if tunnel-digger is up.

SvenRoederer, 2020-07-08:
@kls0e has written something on the Mailinglist recently (https://lists.berlin.freifunk.net/pipermail/berlin/2020-June/040656.html)