freifunk-berlin/firmware #832: LuCI configuration-change: timeout of confirmation triggers rollback

SvenRoederer, 2020-07-19:
### Bug report

**What is the problem?**
Changing Network or olsr parameters via LuCI will not apply, as they get reverted.

**What is the expected behavior?**
A change of these parameters should be applied.
Currently it seems to fail as the new LuCI (OpenWrt-19.07) waits for a confirmation for the client (JS-code) that the changes are woking. There is a 30 seconds timeout for this. A change on some network related parameters causes the restart of the routing, which will very likely take more than 30 seconds to reestablish. 

**Firmware Version:**
I found this with "Dev-Monthly-Master 295051dd"; which is 835bcb9 + #794

everloop2, 2020-08-06:
in recent versions of openwrt seen 60-90sec session timeouts

luci-base: extend rollback timeout to 90 seconds
https://git.openwrt.org/?p=project/luci.git;a=commit;h=81cf99a50706a75a2259296e4aab3f7b3bf5d54e

everloop2, 2020-09-26:
think we can close due: https://github.com/freifunk-berlin/firmware/pull/802 19.07.4 update.

closed 2020-09-26 by everloop2

SvenRoederer, 2020-09-26:
Thanks for watching this. Indeed 90 sec should be enough to restart olsr and re-establish the routing.

A sidenote: the affected node was installed with an image dated before this rollback happened and upgraded to a more recent version. As the config was kept, the old "30 secs timeout" was kept also. 
But as we only take care for such changes between official releases of our firmware, I don't think we need a migration to take care.