freifunk-berlin/firmware #834: OpenWrt-base for next minor-release (year 2020)

SvenRoederer, 2020-08-02:
Currently we are in much the same situation as in #583 two years ago. It seems OpenWrt is trying to branch a 20.x version "soon" and I suggest again to switch to OpenWrt-master in order to prepare the next minor release (which was already targeted in #583).

Again, we have exact the same situation, we had before. So the same arguments for the switch rather soon than later. In addition @PolynomialDivision is actively working on the OpenWrt-master already.

* http://lists.openwrt.org/pipermail/openwrt-devel/2020-July/030593.html
* http://lists.openwrt.org/pipermail/openwrt-devel/2020-August/030606.html

SvenRoederer, 2020-08-06:
Probably it was a bit unspecific in the headline and #583. This is not about using OpenWrt-master on the final released firmware, it's about switching to OpenWrt-master now and adapt the Firmware to work on it. This way we have a chance to have some working code ready, wen OpenWrt 20.x is released.

SvenRoederer, 2020-09-07:
I just updated PR #802 to OpenWrt-19.07.4, but no matter is this will be merged I'll use this point to branch of a OpenWrt-2020.x based series on my personal repo and stop working on 19.07.

PolynomialDivision, 2020-09-19:
Be careful with switch to 5.4 kernel. Many devices have weird bugs and that change to DSA config is not working smoothly.

- https://forum.freifunk.net/t/jffs2-corruption-mit-5-4-kernel/22247
- https://forum.openwrt.org/t/ipq40xx-low-performance-when-sending-traffic

SvenRoederer, 2020-09-19:
> I just switch back to 4.19 kernel for mostly all devices. :/

Which means OpenWrt-19.07? As to my knowledge all target on OpenWrt-master are on Linux 5.4.

PolynomialDivision, 2020-09-19:
> Which means OpenWrt-19.07

We have a 19.07.4 release, but most of the time I use master and change back kernel to 4.19.

SvenRoederer, 2020-09-27:
Wasn't you insisting to use pre-build Openwrt images (Imagebuilder)? How do you accomplish it to use a different kernel?

SvenRoederer, 2020-10-22:
> I just switch back to 4.19 kernel for mostly all devices. :/

As per http://lists.infradead.org/pipermail/openwrt-devel/2020-October/031656.html and recent commits in OpenWrt-master 4.19-support gets removed. So 5.4 will get fixed ... or is probably already (https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=86fdc8abed5992a74078b000b5ff9da723b6f46b).

everloop2, 2020-10-25:
> Be careful with switch to 5.4 kernel. Many devices have weird bugs and that change to DSA config is not working smoothly.
> 
> * https://forum.freifunk.net/t/jffs2-corruption-mit-5-4-kernel/22247
> * https://forum.openwrt.org/t/ipq40xx-low-performance-when-sending-traffic
> 
> I just switch back to 4.19 kernel for mostly all devices. :/

yep 5.4 still a bit buggy - did not test much, will be a maintained release - go for it :)

4.19 got USBhang! regression on some [EasyBox803](https://bugs.openwrt.org/index.php?do=details&task_id=2808&string=lantiq&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=) (not a that important box)

4.14 works like charm for boxes got around (lantiq: xway / xrx200..)

PolynomialDivision, 2020-10-25:
> 4.14 works like charm for boxes got around (lantiq: xway / xrx200..)

Today I realized that old kernel perform so good, since sometimes they ignore everything u give to them. xD
https://github.com/openwrt/openwrt/commit/a181dbc2e3caa044b74566a5884bc58e986ae458

The support is getting better and better. For socs like ipq40xx (avm frit!box) we have to switch to DSA, since with normal swconfig we have very hard limitations and can not support e.g. isolated ports and nested vlans together...

SvenRoederer, 2020-10-25:
5.4 of OpenWrt master will become next release anyway and the most annoying bug "LANport intermittent link" seems fixed (https://lists.berlin.freifunk.net/pipermail/berlin/2020-October/051732.html) I'll cherry-pick from my fork, to PR this change. As mentioned 5.4 support will become better ...

There is always the option to maintain a openwrt-19.07 based branch if somebody will take care.

SvenRoederer, 2021-05-08:
Master is OpenWrt 21.02 in the mean time.

closed 2021-05-08 by SvenRoederer