freifunk-berlin/firmware #837: daemon.warn dnsmasq-dhcp: DHCP packet received on br-wan which has no address

everloop2, 2020-09-26:
**Firmware Version:**

master: Freifunk Berlin Development 91b6ae0 -> think all others too

**What is the problem?**

syslog is spammed by: `daemon.warn dnsmasq-dhcp[2184]: DHCP packet received on br-wan which has no address` after bootup, (dunno if DSL_BOX related?)

**What is the expected behaviour?**

dnsmasq should not care for br-wan / be quiet

**workaround/solution:**

```
ssh root@frei.funk
vi /etc/config/dhcp
```
insert:
```
config dnsmasq
list notinterface 'br-wan'
```

SvenRoederer, 2020-09-26:
I've also seen this messages a lot on regular LAN-based uplinks, so nothing special to "DSL-Boxes".
But will this change break getting dhcp-config on the interface?

SvenRoederer, 2020-09-26:
> But will this change break getting dhcp-config on the interface?

Got it ... it's stopping the dnsmasq dhcpserver to listen on this interface. @everloop2 can you PR this?

SvenRoederer, 2020-09-26:
Btw. I'm quite sure this is caused by the "no-tunnel" images when ffuplink requests an address on this interface too (#624).

Based on this, shouldn't we also include "ffuplink_wan" to the notinterface list?

everloop2, 2020-10-01:
yep, ffuplink_wan also can/should be ignored.

created pull: https://github.com/freifunk-berlin/firmware-packages/pull/212

closed 2020-10-04 by SvenRoederer