freifunk-berlin/firmware #838: xDSL: WAN not configurable via LuCI due missing packages

everloop2, 2020-09-26:
**Firmware Version:**

master: Freifunk Berlin Development 91b6ae0 -> think all others too

**What is the problem?**

boxes got a DSL port (FritzBox 7360x, 3370..) -> WAN not configurable via LuCI due missing packages:

```
luci-proto-ppp
luci-proto-ipv6
ppp
ppp-mod-pppoe

```
**What is the expected behaviour?**

WAN should be configurable via LuCI - expecially to switch: DSL_WAN to LAN_WAN

(OpenWRT stable/snapshot needed packages are not missing)

**Workaround/Solutions:**

direct use of DSL_uplink is not done in most cases and [setting it up](https://openwrt.org/docs/guide-user/network/wan/isp-configurations#general_xdsl_configuration_step_by_step) not k.i.s.s. .

1. set WAN default to DHCP -> prevents WAN is not configurable via LuCI due missing ppp_xx packages:

```
/firmware/openwrt/target/linux/lantiq/base-files/etc/board.d/02_network

ucidef_set_interface_wan "$interface_wan" "dhcp"
```
or set it via Freifunk wizard.

2. remove all xDSL related packages - if DSL_uplink is needed, manually download packages and set it up after wizard

/firmware/packagelists/..

```
# add/enable to prevent Luci DSL_WAN no proto bug
-luci-proto-ppp
-luci-proto-ipv6
-ppp
-ppp-mod-pppoe

# remove lantiq xDSL: xway_ADSL (EasyBox803..) and xrx200_xDSL (FritzBox 7360x, 3370..)
-ppp-mod-pppoa
-kmod-ppp
-linux-atm
-ltq-adsl-app
-kmod-ltq-atm-ar9
-kmod-ltq-adsl-ar9
-kmod-ltq-adsl-ar9-mei
-kmod-ltq-adsl-ar9-fw-a
-kmod-ltq-adsl-ar9-fw-b
-kmod-ltq-atm-danube
-kmod-ltq-adsl-danube
-kmod-ltq-adsl-danube-fw-a
-kmod-ltq-adsl-danube-fw-b
-kmod-ltq-adsl-danube-mei
-br2684ctl
-bspatch
-dsl-vrx200-firmware-xdsl-a
-dsl-vrx200-firmware-xdsl-b-patch
-kmod-atm
-kmod-ltq-atm-vr9
-kmod-ltq-deu-vr9
-kmod-ltq-ifxos
-kmod-ltq-ptm-vr9
-kmod-ltq-vdsl-vr9
-kmod-ltq-vdsl-vr9-mei
-linux-atm
-ltq-vdsl-app
-ltq-vdsl-vr9-vectoring-fw-installer

# Voice over IP (most boxes no support)
-kmod-ltq-tapi
-kmod-ltq-vmmc
```

SvenRoederer, 2020-09-26:
other options would be:
* include the *ppp packages in the default packagelists
* or to add a separate packagelist for "DSL-uplinks"
Both options I don't like really, as they "bloat" the image or add another packagelist for this edge-case.

The 2nd option mentioned by @everloop2 seems also very "bloaty", as it will require to add a lot of excludes the the common packagelists. I assume the listed packages are included by upstream per default, but should not interfere with normal operation (beside of additional RAM-usage). 
As our buildsystem is not supporting board-specific packagelists / package-excludes I'm favour of leaving this installed packages as the are. 

The 1st option (changing the defauit from PPP-uplink to DHCP-uplink) seems the most practical. As this is very likely the most common usecase and the wizard don't support other uplink-types (#601). 

P.S: A prototype to use board specific packages, in addition to the generic packagelists, can be discussed in #765

SvenRoederer, 2020-09-26:
@everloop2 can you post the relevant section of the uci-config?

everloop2, 2020-10-02:
1. set WAN default to DHCP -> would be easy way with less impact on generated images to upstream ones:

disadvantage i've seen so far:
xDSL drivers are loaded and DSL_boxes try to get/search DSL signal (expected behavior) -> results in xDSL connection loop spamming syslog about pppxx and ?wasting some power (cpu / ram..)?

disabling WAN at boot does stop xDSL connection loop

and together with https://github.com/freifunk-berlin/firmware-packages/pull/213 syslog is not spammed by dnsmasq bothering about br-wan and ffuplink_wan

2. remove all xDSL related packages - if DSL_uplink is needed, manually download packages and set it up after wizard:

gluon: lantiq-xrx200: remove modem packages from image #2087: https://github.com/freifunk-gluon/gluon/pull/2087

~1MB less images, no xDSL connection loop, no software to run on blobbed xDSL hardware (only some paranoia / "less is more")

all related packages to exclude can be put bottom in /firmware/packagelists/xxx.txt - if they dont exist for specific targets they will be ignored - "doesnt hurt".

everloop2, 2020-10-25:
> @everloop2 can you post the relevant section of the uci-config?

think this is good place:

https://github.com/freifunk-berlin/firmware-packages/blob/master/defaults/freifunk-berlin-network-defaults/uci-defaults/freifunk-berlin-network-defaults

```
# setup wan as bridge, set wan to dhcp (prevents wan not configurable due missing pppxx package_s)
uci set network.wan.type=bridge
uci set network.wan.proto=dhcp
```

SvenRoederer, 2021-06-27:
PR #863 should fix the issue by adding the required packages for the relevant boxes only.