freifunk-berlin/firmware #844: Failed to execute template 'freifunk/index'

ilippert, 2021-03-18:
### Bug report

**What is the problem?**
I installed the firmware http://buildbot.berlin.freifunk.net/buildbot/unstable/ramips-mt76x8/151/notunnel/freifunk-berlin-next-0743518-glinet_gl-mt300n-v2-sysupgrade.bin from scratch. And get this result on a GL-MT300N-V2:

```
/usr/lib/lua/luci/template.lua:55: Failed to execute template 'freifunk/index'.
A runtime error occurred: /usr/lib/lua/luci/tools/freifunk/assistent/tools.lua:57: attempt to index local 'e' (a nil value)
stack traceback:
	/usr/lib/lua/luci/tools/freifunk/assistent/tools.lua:57: in function 'hasRootPass'
	[string "/usr/lib/lua/luci/view/freifunk/index.htm"]:24: in main chunk
stack traceback:
	[C]: in function 'n'
	/usr/lib/lua/luci/template.lua:55: in function </usr/lib/lua/luci/template.lua:47>
	(tail call): ?
	/usr/lib/lua/luci/dispatcher.lua:844: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:816: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:412: in function </usr/lib/lua/luci/dispatcher.lua:411>
```

Akira25, 2021-03-20:
@ilippert The firmware you've installed is an development image from June 2020. Most of the berlin firmware developers started a new fork of this firmware. We've had the first stable release of that fork around a month ago (-> falter-1.1.0). You can download an image for the GL-MT300N-V2 via the new [Firmware-Selector](https://download-master.berlin.freifunk.net/falter/).

If you find any issues with falter, you might report them in our repository at https://download-master.berlin.freifunk.net/falter/

closed 2021-03-20 by Akira25

ilippert, 2021-03-20:
Thanks, I am not sure this is reflected at https://util.berlin.freifunk.net/hardware?name=glinet_gl-mt300n-v2 or https://wiki.freifunk.net/Berlin:Firmware#Unterst.C3.BCtzte_Hardware ... I am not sure whom to notify re this...

Akira25, 2021-03-20:
At least in the firmware article in the wiki, we changed that. But maybe we should hide that old table and state more clearly, that is was depracated. I will do something on that later on. :)