freifunk-berlin/firmware #855: OWM-App: verschieben ins globale Freifunk-repo

SvenRoederer, 2021-05-15:
Ich denke es macht Sinn die OWM-App komplett ins nach Freifunk/openwrt-packages zu verschieben, da der code ja nicht wirklich Berlin-spezifisch ist.
Ich vermute mal der historische Grund ist das niemand den Code damals ins OpenWrt/LuCI repo schieben wollte und kurze Wege bleiben sollten. Da das Freifunk-repo ja nicht zur OpenWrt-organisation gehört, sollten die kurzen Wege bestehen bleiben, die App aber weitere Verbreitung finden.

Akira25, 2021-05-16:
Ich würde beim derzeitigen Zustand der App dringends davon abraten, diese weiter zu verbreiten. Dies hat mehrere Gründe:

* In ihrem derzeitigen Zustand im [freifunk-packages-repo](https://github.com/freifunk-berlin/firmware-packages/tree/master/utils/luci-app-owm) ist die App sicherheitstechnisch kaum tragbar. Ich glaube kaum, dass der reguläre Freifunk-Nutzer erwartet, dass wir seine MAC-Adresse in eine zentrale, öffentlich ansprechbare Datenbank kippen. Das widerspricht den Werten der Freifunk-Community in Berlin eklatant.
* OpenWrt wird perspektivisch die Lua-Unterstützung einstellen. Eine Portierung in eine andere Sprache ist also sinnvoll.

Eine wesentlich weniger geschwätzige Version von `owm.lua` kannst du im [falter-repo](https://github.com/Freifunk-Spalter/packages/tree/master/luci/luci-app-falter-owm) finden. Eine Portierung nach `sh` ist bereits [in Arbeit](https://github.com/Freifunk-Spalter/packages/issues/159).

SvenRoederer, 2021-05-16:
@Akira25 PRs for https://github.com/freifunk-berlin/firmware/issues/585 sind willkommen, wurden aber noch nicht gesichtet. Eine Neuimplementierung wurde von @booo auch angeregt (https://github.com/freifunk-berlin/firmware/issues/819#issuecomment-655435191).

Das OWM-paket ist aber nicht nur der CLI-client sondern auch die LuCI-app, daher gibt es schon Sinn

SvenRoederer, 2021-05-30:
https://github.com/freifunk/openwrt-packages/pull/46#issuecomment-850910823

closed 2021-05-30 by SvenRoederer