freifunk-berlin/firmware #862: From email: wizard cannot be run anymore after initial installation

ilippert, 2021-06-18:
Using falter v1.1.0 - 0968abf ath79 - generic

as example: after the first time installation of falter, the wizard cannot be run again. This is unfortunate, because (in earlier versions) it allows users to reconfigure the device without having to go through the shell.

---

Date: Thu, 17 Jun 2021 17:51:36 +0200 (CEST)
From: Malte <freifunk@antenne.yagii.de>
To: wirelesslan in Berlin <berlin@berlin.freifunk.net>
Subject: Re: [Berlin-wireless]  nachträglich  Mesh-IP-Adressen
	ändern
Message-ID:
	<alpine.DEB.2.21.2106171750110.21354@bluebox.antenne.yagii.de>
Content-Type: text/plain; charset="utf-8"; Format="flowed"

On Thu, 17 Jun 2021, Rainer-M. Fritsch - Listen wrote:

> wie kann ich in einem Berliner Freifunkrouter nachträglich die Mesh-IPs 
> und das DHCP-Network ändern? In welcher Konfigurationsdatei sind diese 
> eingetragen?

Eigentlich sollte das gehen, indem man den Wizard einfach nochmal 
bis zum Schluss durchgeht und die Werte ändert.

Falls das nicht so funktioniert, wäre das ein Bug (denn es ging wenigstens 
vor ein paar Jahren so).

Grüße,
Malte

pmelange, 2021-06-19:
reference: https://github.com/freifunk-berlin/firmware/issues/661

SvenRoederer, 2021-06-19:
As this issue is essentially a duplicate of #661 I like to close. 
It's also referring to foreign @freifunk-spalter project, which we don't track here. As Freifunk-Spalter started with our codebase it's technically not that wrong here, anyways.

closed 2021-06-19 by ilippert